RobertLorincz/Smart-Orbiter-SO3 #1: Spare parts?

jjl40, 2024-09-03:
The wires for the hotend cooling fan are too short to reach the tool board and are likely to get pinched. If the wires were 10 mm longer it would be much easier to install. The next problem is finding a replacement.

RobertLorincz, 2024-09-08:
What is the length of your wire, in my case, I had not really issues, but could it be that yours came shorter?

RobertLorincz, 2024-09-08:
For spare parts, you need to ask your local supplier, I insisted a lot on having spare parts also available, but for some reason vendors are not open to it. Its a real issue, I admit, but unfortunately it's hard on my side to change this. It would be better if customers complained directly to the vendors because they can order replacement parts from LDO...but they are not willing to right now. I even asked to at least provide spare parts via AliExpress...so far, still no parts there. Fabreko and Filastruder has some in the US but that is not convenient for everybody.

jjl40, 2024-09-08:
Thank you for your prompt reply Robert,

The hotend fan wire is ~120mm. My mount is a bit unusual due to its layout. I fixed the problem by creating a conduit in the mount so the wire has a direct route to the toolboard. [image0.jpeg]

The only other issue I have is that I have not found a description of how to use the filament load/change. Everything is working fine (the assembly and installation is well documented) but I have not used the automatic loading/change feature.

Cheers

John Lytle

jjl40, 2024-09-08:
Thanks again Robert,

I contacted KB3D, where I purchased my Orbiter 3, and Kris informed me that he has the fans in stock. The website was not updated at that time.
This is very convenient for me as KB3D is a short drive for me, just up the road in Twinsburg, Ohio.

Cheers

John Lytle