RobertLorincz/Smart-Orbiter-SO3 #2: revo vs non-revo

mindforger, 2024-09-16:
I have no idea what the difference between the revo and non-revo version is, in the mode the only difference seems to be the cableing of the heater, can someone explain? because shop sites also do not state the difference clearly only a 20is dollar difference

CarVac, 2024-09-23:
Revo versions are for the E3D Revo quick-swap integrated-heatbreak nozzles, not v6 nozzles.

I'd expect that the heatbreak connection is different, the heater is going to be a revo heater of some sort (what wattage?), and there'd be some sort of attachment point for the heater's spring.

mindforger, 2024-09-23:
thank you, i did not visit the site, i was browsing github for extruder options since i wanted to build my own head for my sv08 ... unfortunately i already ruled out the orbiter v3.0 becasue it is too wide to replace the original head and i was unsure if i could use it with the trianglelabs chc pro ... it was a bit confusing to only see the difference in the 3d file

closed 2024-09-23 by mindforger