RobertLorincz/Smart-Orbiter-SO3 #3: Wire colors in SO3.cfg are backwards

CarVac, 2024-09-21:
I tried to wire my SO3 based on the config file but was getting nonsensical results.

I checked for the thermistor and realized that all the SO3 integrated board wires had backwards colors.

Here's my findings:

```
Orange: ground
White:  5v
Gray:   RGB
Brown:  filament sensor
Yellow: unload button
Red:    temperature
Blue:   hotend light
```

Thus the aliases code should be:

```
  filament_sensor_pin = PC7,         # BROWN signal shall be connected to a digital I/O pin
  filament_unload_pin = PC6,         # YELLOW signal shall be connected to a digital I/O pin
  RGB_pin = PG8,                     # GRAY signal shall be connected to a digital I/O pin
  extruder_temperature_pin = PC1,    # RED signal shall be connected to a temperature sensing input pin
  hotend_lit_pin = PA1,              # BLUE signal shall be connected to a digital I/O pin
```

Also the colors should be documented somewhere else. Maybe on the schematic.

nico9889, 2024-09-21:
I can confirm the issue on the Revo version of the Smart Orbiter, I was convinced the IC board was faulty but after trying @CarVac's color coding I was able to lit the NeoPixel using an Arduino (I still need to test everything else since I need to crimp again all the connectors... and explain to the vendor that the board is probably magically working again and I don't need a replacement anymore 🙈).

@RobertLorincz I'd suggest to create a dedicated page with instructions to setup the Smart Orbiter, I completely missed the reference image under the point "3.3 Integrated Smart Features" in the current page (https://www.orbiterprojects.com/so3/) since it contains **a lot** of mixed information and I didn't expect to find wiring related information under the "Firmware configuration" section.

[image]

I think the image in the current page is fine, but it need to be put in a more evident place. I wouldn't rely on the cable colors unless there's some kind of warranty that they won't change in the future (I.E: due to a manufacturer decision).

CarVac, 2024-09-21:
Huh, I didn't register that image at all.

Maybe you're right, it probably shouldn't specify wire colors in case it changes.

But I'm glad my findings were able to help someone. Now to wait for my replacement hotend to come in... I had a DOA CHC.

nico9889, 2024-09-21:
I created a pull request to fix the colors specified in the configuration and I added more instruction to wire the Smart Orbiter as well to avoid any possible future mistake. I took the liberty to add you as a co-author since you found both the error and the right wire colors.

RobertLorincz, 2024-09-23:
The problem is that the supplier mixed up the colors and there is not only one variant out there. So I prefer people follows the pin numberings and definition instead of the wire colors.

nico9889, 2024-09-23:
That's exactly what I was worried about. I don't know how familiar you are with GitHub, but I opened a pull request #4 to update the comments in the configuration. I changed the colors according to what CarVac reported and added the pin numbering on the top. Should I remove the colors? Is it fine for you to have the pin numbering inside the configuration? Do you prefer updating the configuration by yourself instead?

If you'll find some time to do that, I'd still strongly suggest to create a dedicated web page with the instruction to setup the Smart Orbiter 3, it's easy to miss the image with the pin numbering on the current page. As a side note I need to make you notice that the website is completely missing information about the Revo version (aside from the small paragraph in the blog post about the SO3), so some people are getting confused #2.

The configuration need to be updated for sure since it's misleading, but about the website it's just a my opinion.

CarVac, 2024-09-23:
I would want to remove all references to colors in the configuration, then, and just have a physical order description in the file.

nico9889, 2024-09-23:
I agree, but I’ll wait for RobertLorincz final decision since this is it’s own project.

closed 2025-01-16 by RobertLorincz