RobertLorincz/Smart-Orbiter-SO3 #7: Friction in the upper filament path

tamebits, 2024-11-30:
I just received my SO3. It feels like there is too much friction in the first section. The force required to get the filament moving is somewhere in the range of 7-10 N.
I see something in the filament path. I guess it is some actuator for a micro switch that serves as a filament runout sensor.

Is this amount of friction normal?

RobertLorincz, 2024-12-01:
No, it's not normal; the filament should go in pretty easily. Please take the extruder apart and check if there is not some debris inside the extruder. ALso check the position of the internal electronics board; it could be that the board is not in the correct place and it is closer to the filament than it should be.

tamebits, 2024-12-01:
I took it apart and did not find any debris. I checked the filament diameter and it is 1.75 mm.
The internal board sits flush on the aluminium part. SW2 sits flush on the PCB.
PCB + SW2 fully compressed is 5mm thick. PCB + SW2 released ~ 5.5 mm
The actuator ball is 4 mm.
Anything wrong with that?

The ball left an indent in the switch's button. Looks like the switch is being "over-actuated" which results a lot of friction.

[image]

RobertLorincz, 2024-12-03:
I can confirm the issue; I'm trying to find out which component is out of spek. It seems there is an issue with some parts causing this issue, I'm in contact with LDO to find and correct the issue. Sure, you will receive replacement parts to fix this.

tamebits, 2024-12-19:
Thank you Robert,

I asked LDO to get in touch with me. Looks like I have to find another project for the holidays. Or maybe I'll just print a mockup extruder for now ;-)
Merry Christmas and a Happy New Year to you!

tamebits, 2025-01-18:
LDO sent me a replacement part for the upper filament path.

They say the fixed roundness and taper of the ball holes.
Unfortunately that didn't change anything for me. I still have to push with about 8 N.
I guess the steel ball is till over-actuating the microswitch.

msticninja, 2025-01-23:
I just received 4 SO3 Revo Editions, all of them have this problem. It's nearly impossible to push filament in by hand. Also, there were two mounting holes that weren't even drilled and tapped on one of them. What's going on with manufacturing?

RobertLorincz, 2025-02-03:
I ask LDO to look into this issue. its OK to have slight friction, but it should go in easily.

RobertLorincz, 2025-02-03:
Would it be possible to send me pictures with the sensor switch position? Maybe is not flush with the PCB and there is a gap between the sensor swicth and the PCB?

tamebits, 2025-02-18:
Sorry for the late resposne, I was too busy with other things.

The button is slightly rotated, but it sits flush with the board. The board is 1.2 mm thick.
PCB plus released button is 5.5 mm
PCB plus fully pressed button is 5 mm

[image]

[image]

[image]

The PCB barely sits flush on the CNC part.

[image]

I think the switch is too close to the filament path.

Rico-origin, 2025-03-24:
I received my SO3 on the 14.3.2025, and this problem still persists. I purchased it from 3DJake in Europe. Is there a solution for this?
As my printer still uses Marlin 2 firmware, and there is no Marlin 2 configuration for this extruder and my printer combo, I am not using any smart features for now, so I just dissasembled the extruder and removed the ball, which actuates the switch. 
My switch doesn't have an indent from the ball as on the picture above.
Please let me now what is the fix and if I need do contact the distributor. Maybe slightly smalled diameter ball for switch actuation?

tamebits, 2025-03-24:
As I didn't hear form LDO in a while so I decided to fix the problem myself.

I carefully reduced the button's height until the friction felt ok. After that I checked that the steel ball still actuates the button.

I'm sure you can also increase the diameter of the filament bore but I figured replacing the SMD button would be easier than replacing the CNC part in case I mess things up.

tamebits, 2025-03-24:
>  Maybe slightly smaller diameter ball for switch actuation?

I only had 3.5 mm around which is already too small to actuate the button

If you find balls in 0.1 or even 0.05 mm increments you could test which one is good for you.
It probably doesn't take much to get rid of the friction.

RobertLorincz, 2025-03-24:
Thanks for the feedback, I will ask again LDO guys to double check this before shipping. The filament should go inside relatively easily. It's not OK to have high friction in there. One easy way to recuce this would be to use a smaller steel ball or try to cut a little down from the top of the button with a sharp cutter knife.

MerlinBaumann, 2025-04-30:
Hi
While my extruder was fine at first, the friction issue appeared after around 50h-100h of printing the first time. When the tensioning lever is removed, I saw that the main drive gear was displaced axially to the front (fan). After realigning the teeth for the filament with the bore, the extrusion was good again. After some more printing hours the flow rate of PETG with a CHT 0.4mm nozzle decreased from more than 18mm3/s to less than 10mm3/s before the extruder starts to skip steps.
Filament changes need a lot of force, too.

Removing the tension lever again, shows that the main drive gear moved axially to the back this time. 
(The set screw was in place and tight during operation)

[image]
[image]

The scratched off anodization at the top of the upper edges of the filament guide idicates, that the filamet did rub against the guide. Sometimes scratch maks chould be seen on the side of the filament, too. 

The secondary dive gear is axailly floating on the bearing in the lever only guided by the filament. 
Thus the axial displacement of the main drive gear results in the same displacement of the secondary drive gear. 
The picture below shows scratched anodization on the lever on the left side. This further increases friction as aluminum is not a metal that slides well.

[image]

I have no explanation for the scratches in the middle of the gear.
Maybe the indents of the screw on the pivot point of the lever change the geometry to allow for contact with the filament guide. Maybe not. I didn't measure the details in the CAD model. Maybe it was some debris at some point.

[image]

However, it seems to me that main drive gear can shift axially on its shaft, which leads to friction in multiple places. Is there a way to constrain axial movement of the main drive gear on the shaft? The set screw and pressure fit don't seem to provide sufficient force to do so.
Could the sum of all tolerances over the shaft and housing parts result in to much axial clearance for some units while others good?

What do you think?

msticninja, 2025-07-01:
I removed .55mm from the tops of the switches to get reliable triggering with very little friction. Be careful with this route, as taking off too much will cause unpredictable/incorrect runout alarms to trigger.

b-A-shi, 2025-07-03:
That felt very wrong, but I managed to solve it. 
Instead of cutting the button, I used a fine sanding paper to lower the button

Lebensgefahr, 2025-10-30:
I also have the same issue with the ball and the filament feeding resistance :-( I thought it was supposed to be that way.

Lebensgefahr, 2025-10-30:
> The issue is the housing itself. The little support ledge near the button is around 0.5mm taller than the height of the mounting points for the PCB. This results in the PCB sitting unevenly. Since that little ledge is closest to where the button is, this means that the button sits 0.5mm closer to the filament path than it should be.
> 
> [image]

I carefully measured this platform with a digital caliper, as precisely as this tool allows. I found it to be 0.2 mm higher — definitely not 0.5 mm. There is a slight tilt of the PCB, barely noticeable to the eye. The 0.2 mm could be carefully scraped off with a flat screwdriver, but confirmation from the manufacturer or other users is needed to ensure that 0.2 mm is the correct height. Especially considering that the board rests on its small edge. On my unit, the button is triggered by filament around 1.5 mm thick — it doesn’t seem to trigger at 1.4 mm.